robinson/gos7 #1: Example of AG Multiple Read

rtmaster, 2018-03-16:
Great project @robinson !
Can you please post an example of Multiple Read?
Thanks

robinson, 2018-03-19:
Hi @rtmaster , please take a look into this file: https://github.com/robinson/gos7/blob/master/test/client.go
at function: ClientAGReadMulti, there you can see multiple read functions. 
Cheers, R

sandeep51989, 2018-03-21:
@robinson:  
Thanks for the project.
 I am having trouble  getting the multi read of multiple DB's .Please find my code below. 

Multi Read :

func CleintAGReadMulti(client gos7.Client) {
	data1 := make([]byte, 1024)
	data2 := make([]byte, 1024)
	data3 := make([]byte, 1024)
	var error1, error2, error3 string
	//var error1 string

	var items = []gos7.S7DataItem{
		gos7.S7DataItem{
			Area:     0x84,
			WordLen:  0x06,
			DBNumber: 23,
			Start:    0,
			Amount:   256,
			Data:     data1,
			Error:    error1,
		},
		gos7.S7DataItem{
			Area: 0x84,
			WordLen:  0x02,
			DBNumber: 14,
			Start:    0,
			Amount:   512,
			Data:     data2,
			Error:    error2,
		},
		gos7.S7DataItem{
			Area: 0x84,
			WordLen:  0x02,
			DBNumber: 20,
			Start:    0,
			Amount:   512,
			Data:     data3,
			Error:    error3,
		},
	}
	err := client.AGReadMulti(items, 3)
	if err != nil {
		fmt.Println("Error", err)
	}
	var s7 gos7.Helper
	var value1, value2, value3 float32
s7.GetValueAt(data1, 168, &value1)

Value1=0.0

Individual DB Read:

err := client.AGReadDB(address, start, size, buf)
	if err != nil {
		log.Fatal(err)
	}
	// result := binary.BigEndian.Uint16(results)
	var s7 gos7.Helper
	var result float32
	s7.GetValueAt(buf, 168, &result)

Result =1761.5

I tried using multipe word length if that gives me a issue. I tried but whenever multi read is 0 value whereas  the individual read does the trick for me. But i dont want to bombard the PCL with hitting a request for each DB. 

IF  i am doing any thing wrong please correct me 

Thanks

robinson, 2018-03-21:
@sandeep51989
what does the number 168 mean?

sandeep51989, 2018-03-21:
Its just the position of a value in byte array.

robinson, 2018-03-21:
Lets use 0 instead of 168

rtmaster, 2018-03-26:
Please @robinson can you help with the below practical example?

In DB3 I have:
- Offset 0 --> real
- Offset 8 --> int
- Offset 10 --> bool

In DB2 I have:
- Offset 0 --> real
- Offset 4 --> real
- Offset 8 --> real

How would you read and parse that with a single call to ClientAGReadMulti?

closed 2021-05-17 by robinson