robinson/gos7 #2: Read boolean

rtmaster, 2018-03-16:
please can you advise if there is a function to read booleans?
@robinson

robinson, 2018-03-19:
@rtmaster read boolean is similar with read any types of data:
buf := make([]byte, 255)
	err := client.AGReadDB(address, start, size, buf)
	if err != nil {
		t.Fatal(err)
	}
here you got the the buf, which is an array of byte, then just get the value from buf:
var s7 gos7.Helper
var result boolean
s7.GetValueAt(buf, 0, &result)
Now you got the result is a boolean.
Cheers, R

sandeep51989, 2018-03-21:
I see there to be a problem with this approach. If the boolean is given in a byte it works. How can we have get booleans at the bit position in a byte. For example if i want to get the boolean value at 180 the position at bit 6. 

According to examples given. 
s7.GetValueAt(buf, 180, &result)

All the function does is read the record at position byte 180 but how do we get the value at the bit position off the byte.

pizzalord22, 2019-07-16:
sorry for reviving this old thread but when i read a byte containing 8 bool bits it returns 1 byte in the buffer array instead of 1 byte for each bool value. so i can not get individual bool values

petar-dambovaliev, 2019-10-04:
@pizzalord22 
This will give you the specific boolean you want from the byte.
`myBool = (b & (1 << bitPos-1)) != 0;`

petar-dambovaliev, 2019-10-05:
I have added the boolean support in the helper.
https://github.com/robinson/gos7/pull/17

closed 2021-05-17 by robinson

sunblack110, 2021-06-30:
It's good , but how about write a boolean to PLC, do we need to read it in a byte first, then change a bit in the byte, then we write to PLC , it's not a good idea.