robinson/gos7 #10: Question: S7-200

localleon, 2019-05-20:
Hey man, 
i really want to use your awesome work... Im running an Siemens S7-200 with an CP243-1 as ethernet module and my end goal is to set a marker and therefor control an output. simple stuff...

Now to my problem. The connection handler asks for an rack and slot number which i don't seem to find. Do you have any tips for finding this number? Is it in my Project or CPU defined?

Any help is appreciated. Thx..

robinson, 2019-05-21:
Hi there,

I have not tested with S7-200 but as far as I know S7-200 has no rack/slot . Therefore just put 0/0. Try to take some test and let me know the result. If you have any improvement please feel free to sharing. Cheers, R

localleon, 2019-05-21:
Hi, 
thanks for your quick response. I already tried rack 0 and slot 0. Also found the TSAP (maybe this has to do something with the rack/slot numbers?) of the CP-243-1 in the Project which is 10.06 and tried 10 as rack and 6 as slot. Didn't worked too. 

Client --> ME --> 192.168.8.25
S7-200 --> SPS --> 192.168.178.252
PS: There's a Router between these two Subnets... There should be no problem with this. If i go directly into the subnet 178 the result is the same

It's really difficult to find something about the S7-200, that's probably because it's one of the more "low-end" models. I tried taking some packet captures of Step7-MicroWin talking to the SPS and compared some Packets against each other to figure out if there's some kind of rack/slot id. 

My End Goal is to write to a Marker. I did this in Step7-MicroWin and captured this Packet via Wireshark. This Packet sets Merker M 2.4 to 00
```
2	10.362060755	192.168.8.204	192.168.178.252	S7COMM	91	10.362060755  | ROSCTR:[Job     ] Function:[Write Var]

0000   b4 fb e4 53 a4 4c 08 00 27 cb 58 53 08 00 45 00
0010   00 4d 03 00 40 00 80 06 ba 91 c0 a8 08 cc c0 a8
0020   b2 fc 04 2d 00 66 c5 cf 13 66 aa 4c e6 44 50 18
0030   f8 9c 4a a3 00 00 03 00 00 25 02 f0 80 32 01 00
0040   00 02 00 00 0e 00 06 05 01 12 0a 10 01 00 01 00
0050   00 83 00 00 14 00 03 00 01 00 00

----- WIRESHARK TRANSLATION VIA S7COMM PLUGIN ----
S7 Communication
    Header: (Job)
        Protocol Id: 0x32
        ROSCTR: Job (1)
        Redundancy Identification (Reserved): 0x0000
        Protocol Data Unit Reference: 512
        Parameter length: 14
        Data length: 6
    Parameter: (Write Var)
        Function: Write Var (0x05)
        Item count: 1
        Item [1]: (M 2.4 BIT 1)
            Variable specification: 0x12
            Length of following address specification: 10
            Syntax Id: S7ANY (0x10)
            Transport size: BIT (1)
            Length: 1
            DB number: 0
            Area: Flags (M) (0x83)
            Address: 0x000014
                .... .000 0000 0000 0001 0... = Byte Address: 2
                .... .... .... .... .... .100 = Bit Address: 4
    Data
        Item [1]: (Reserved)
            Return code: Reserved (0x00)
            Transport size: BIT (0x03)
            Length: 1
            Data: 00
```

My first try was to simply get CPU Infos because i thought that should be very straight forward. But even this seems to failing. i can't make sense of these Error Codes... Is EOF end of file?. 
```
package main

import (
	"fmt"
	"log"
	"net"
	"os"
	"time"
	"github.com/robinson/gos7"
)

const (
	tcpDevice = "192.168.178.252"
	rack      = 10
	slot      = 0
)

func main() {
	CPUstatus()
}

func CPUstatus() {
	// TCPClient
	handler := gos7.NewTCPClientHandler(tcpDevice, rack, slot)
	handler.Timeout = 10 * time.Second
	handler.IdleTimeout = 10 * time.Second
	handler.Logger = log.New(os.Stdout, "tcp: ", log.LstdFlags)
	// Connect manually so that multiple requests are handled in one connection session
	handler.Connect()
	defer handler.Close()
	//init client
	client := gos7.NewClient(handler)
	// Get CPU Info
	info, err := client.GetCPUInfo()
	if err != nil {
		log.Println("client request err:", err)
	}
	log.Println("client request data:", info)
}

----- TERMINAL OUTPUT AFTER RUNNING ABOVE CODE
└────╼ go run connection-test.go
tcp: 2019/05/21 19:54:29 s7: sending 03 00 00 16 11 e0 00 00 00 01 00 c0 01 0a c1 02 01 00 c2 02 02 40
2019/05/21 19:54:30 *errors.errorString EOF
tcp: 2019/05/21 19:54:30 s7: sending 03 00 00 21 02 f0 80 32 07 00 00 00 01 00 08 00 08 00 01 12 04 11 44 01 00 ff 09 00 04 00 1c 00 00
2019/05/21 19:54:30 *errors.errorString EOF
2019/05/21 19:54:30 client request err: s7: response data is empty
2019/05/21 19:54:30 client request data: {    }
```
This is the Packet generated by the above Code in Wireshark
```
-- Should get CPU Info -- 
0000   b4 fb e4 53 a4 4c 00 50 b6 28 2f f7 08 00 45 00
0010   00 49 74 33 40 00 40 06 8a 15 c0 a8 08 19 c0 a8
0020   b2 fc cf c0 00 66 d0 58 6b 35 b2 58 ad bf 50 18
0030   01 f6 3c a2 00 00 03 00 00 21 02 f0 80 32 07 00
0040   00 00 01 00 08 00 08 00 01 12 04 11 44 01 00 ff
0050   09 00 04 00 1c 00 00

-- WIRESHARK TRANSLATION ---
S7 Communication
    Header: (Userdata)
        Protocol Id: 0x32
        ROSCTR: Userdata (7)
        Redundancy Identification (Reserved): 0x0000
        Protocol Data Unit Reference: 1
        Parameter length: 8
        Data length: 8
    Parameter: (Request) ->(CPU functions) ->(Read SZL)
        Parameter head: 0x000112
        Parameter length: 4
        Method (Request/Response): Req (0x11)
        0100 .... = Type: Request (4)
        .... 0100 = Function group: CPU functions (4)
        Subfunction: Read SZL (1)
        Sequence number: 0
    Data (SZL-ID: 0x001c, Index: 0x0000)
        Return code: Success (0xff)
        Transport size: OCTET STRING (0x09)
        Length: 4
        SZL-ID: 0x001c, Diagnostic type: CPU, Number of the partial list extract: Identification of all components, Number of the partial list: Component Identification
            0000 .... .... .... = Diagnostic type: CPU (0x0)
            0000 0000 0001 1100 = Number of the partial list extract: Identification of all components (0x001c)
            .... .... 0001 1100 = Number of the partial list: Component Identification (0x1c)
        SZL-Index: 0x0000
```
This seems to succeed but after 2 ACK from the SPS the SPS resets the TCP Connection... even if i stop calling `defer handler.Close()` in my programm

I really appreciate the help and the work you are doing. I would be super nice if i could get this working.

localleon, 2019-05-21:
Okay, so i just found my TSAP in a capture of the COTP Protocol. I currently just hardcoded it into the source to fix my issue. If i am able to resolve/understand the root cause i will provide you with more information and maybe an pull request. 

Could you share some Sample Code for setting Merkers? My Endgoal is to set Merker 2.4 to 1. Some Code would really help me out because im now just fiddling through the function AGWriteMB to find this out.

localleon, 2019-05-21:
The S7-200 is answering to both CPU and PLC Status Functions with an Error Code (CPU : Function not available 0). So these functions are maybe not supported on this model

localleon, 2019-05-25:
So the S7-200 wants a bit instead of a byte for their data. In your libary the datatype is hardcoded to byte. Maybe we could introduce a parameter for AGWriteMB to change the Type. 

For my problems, i just hardcoded everything to use bits. Thanks for your work

closed 2019-05-25 by localleon