robinson/gos7 #15: GetCPUInfo:wsarecv: An existing connection was forcibly closed by the remote host.

lcp0578, 2019-09-02:
main.go

```
const (
		tcpDevice = "192.168.1.110"
		rack      = 0
		slot      = 2
	)
	// TCPClient
	handler := gos7.NewTCPClientHandler(tcpDevice, rack, slot)
	handler.Timeout = 200 * time.Second
	handler.IdleTimeout = 200 * time.Second
	handler.Logger = log.New(os.Stdout, "tcp: ", log.LstdFlags)
	// Connect manually so that multiple requests are handled in one connection session
	handler.Connect()
	defer handler.Close()
	//init client
	client := gos7.NewClient(handler)
	//cpu, err := client.GetCPUInfo()
	//fmt.Println(cpu)
	//fmt.Println(err)
	cpi, err := client.GetCPInfo()
	fmt.Println(cpi)
	fmt.Println(err)
```
run main.go and output:
```
$ go run main.go
tcp: 2019/09/02 12:15:03 s7: sending 03 00 00 16 11 e0 00 00 00 01 00 c0 01 0a c1 02 01 00 c2 02 01 02
2019/09/02 12:15:04 *net.OpError read tcp 192.168.1.89:65393->192.168.1.110:102: wsarecv: An existing connection was forcibly closed by the remote host.
tcp: 2019/09/02 12:15:04 s7: sending 03 00 00 21 02 f0 80 32 07 00 00 00 01 00 08 00 08 00 01 12 04 11 44 01 00 ff 09 00 04 01 31 00 00
{0 0 0 0}
s7: response data is empty
```

769139671, 2019-09-03:
the connection was closed by your PLC

lcp0578, 2019-09-03:
thanks，769139671 is you QQ？

769139671, 2019-09-03:
chinese? 是我qq号

lcp0578, 2019-09-03:
嗯嗯，已加，交流一下

769139671, 2019-09-03:
@lcp0578 没收到，你qq多少

lcp0578, 2019-09-03:
@769139671 577049355

closed 2019-09-10 by lcp0578