robinson/gos7 #42: About Set/Get PLC time function

whitetiger21022014, 2020-12-24:
Hi Robinson,
PGClock**Write**  (implement **Get**PLCDateTime) , is it a mistake between 2 functions?

//implement GetPLCDateTime
func (mb *client) PGClockWrite() (datetime time.Time, err error) 

//implement SetPLCDateTime
func (mb *client) PGClockRead(datetime time.Time) (err error) {

zhangzijie-820, 2021-07-14:
hi when you use PGClockRead to set time of plc，do you produce  a panic?

arhray, 2023-12-22:
I believe there are multiple issue in datetime.go
PGClockRead / PGClockWrite function names are reversed - just confusing
mb.send does not catch error and so will panic if PLC is disconnected
s7.SetDateTimeAt(requestData, 32, datetime).  I had to change this to s7.SetDateTimeAt(requestData, 31, datetime) to format date correctly.

closed 2026-11-16 by robinson