robinson/gos7 #44: Can I connect to Profinet using this library?

Mendes11, 2021-05-13:
Sry if I might be misunderstanding some basic concepts, but I need to read devices data from a Profinet Network in my application and wanted to know if this library can do that!?

I keep seeing things related to S7 protocol mixed in my searches for Profinet libraries so I'm getting a little confused.

robinson, 2021-05-17:
Hi @Mendes11, sorry for the lately response, 
The S7 protocol in this library can be used for communication via PROFINET connector in the PLC device or in add-on communication modules.
BR,
Robinson

Mendes11, 2021-05-17:
I see, after reading a lot I think I got the difference, Thanks! 

This library uses the Profinet connector, but not actually talks with other PLC vendors using Profinet protocol.. (right? haha)

Anyway, thanks for the response!

closed 2021-05-17 by Mendes11