robinson/gos7 #48: Problems with cross compiling for darwin (serial.go)

stefanpfahler, 2021-06-28:
Hey robinson,
first of all thank you for this package!

I am currently using your s7 implementation in one of my projects which requires me to cross compile from a linux machine to darwin. I have a lot of pain doing so because of some package importing cgo dependencies, namely "github.com/tarm/serial". 
I saw you are using this package inside https://github.com/robinson/gos7/blob/master/serial.go, but the contents of this file aren't used anywhere else in this package and also don't seem to be usable for anyone using gos7.

So my question is: is serial.go crucial for this s7 implementation or could it be removed without causing any hidden errors?

robinson, 2021-06-28:
Hi Stefan, thanks for a very good question, I am intending to compile it in M1 chip in next few day when I have free time and my concerning is serial port library, and now you got the problem. Actually you can remove serial.go without any isssues.
VG, Robinson

stefanpfahler, 2021-06-28:
Thanks for this quick answer! Okay.. I'll fork your repository then and remove the serial part there.. But I'd prefer depending on your originial repository in the future. Do you plan on removing the serial part some day or maybe implement it without a cgo dependent package?

robinson, 2021-06-30:
Hi Stefan, more than happy if you could remove it, take some test and build under darwin then make a PR for a branch without serial part. VG, Robinson