robinson/gos7 #49: About write bit to DB addres

sunblack110, 2021-06-30:
Hi Robinson,

I have a small question, as I would like to write a bit to PLC , but seems that I didn't see any function which is able to fit my requirement. 

But I saw that we have the bit type for write, but the method is private.

`
//writeArea write generic area into PLC with following parameters:
//1.area: s7areape/s7areapa/s7areamk/s7areadb/s7areact/s7areatm
//2.dbnumber: specify dbnumber, to use in write DB area, otherwise = 0
//3.start: start of the address
//4.amount: amount of the address
//5.wordlen: bit/byte/word/dword/real/counter/timer
//6.buffer: a byte array input for writing
func (mb *client) writeArea(area int, dbnumber int, start int, amount int, wordlen int, buffer []byte) (err error) {
	var address, numElements, maxElements, totElements, dataSize, isoSize, length int
`

5.wordlen: bit/byte/word/dword/real/counter/timer 

I would like to know why don't we export the function or create a WriteBit function?

For this case, I need to work around it to read the byte first , then change one bit , and write back , but it's not good way as it will take more time to do it .

Do you have any plan to fix this issue or have any other idea about this ?

Thanks,
Han

robinson, 2021-06-30:
Hi Han,
in fact writing bit is a set Boolean function, you can refer at:
https://github.com/robinson/gos7/blob/fa6e7695589584666f3a29644c4586c7f597749c/helper.go#L323
Br, Robinson

sunblack110, 2021-06-30:
Hi robinson,

Thanks for you quickly response . I noticed this function you mean. But this is a only a memory operation to avoid change the old values in this byte, but we need to read the old value first from PLC . I think it's not good way to do it .

Not sure if I make myself clear or not.

Thanks,
Han

cookedsteak, 2022-04-04:
@sunblack110 Here is the solution, you can add  a Method such as
```go
func (mb *client) AGWriteDBBit(dbNumber int, start int, size int, buffer []byte) (err error) {
	return mb.writeArea(s7areadb, dbNumber, start, size, s7wlbit, buffer)
}
```
Just change the parameter `s7wlbyte` into `s7wlbit`.

jwzl, 2023-12-22:
In reality, it seems that s7wlbit is not work and just read and write a byte is okay for change a bit.