robinson/gos7 #51: How do I connect to s7-1200？There are no tracks and slots in s7-1200，errors：invalid pdu received or invalid buffer passed to send/receive '19'

XQacc, 2021-11-01:


robinson, 2021-11-02:
Hi,
for s7-1200 just add slot=0, rack=0 and it should work.
BR

closed 2021-11-02 by robinson

XQacc, 2021-11-03:
> Hi, for s7-1200 just add slot=0, rack=0 and it should work. BR

set slot=0, rack=0 ，errors：client.AGReadDB: ISO : Invalid Buffer passed to Send/Receive'19'
and used the packet capture tool to view protocol is ISO 8073/COTP，is need other setting？

robinson, 2021-11-04:
no, I think the configuration is that, but you are reading the wrong size of data... i suggest you take a look in the example

iotopo, 2022-03-15:
@XQacc you may got a 8104 error code，hope help you：[https://support.inductiveautomation.com/hc/en-us/articles/4414608266893-Error-reading-S7ReadResponse-Error-code-8104](url)