robinson/gos7 #52: When will support PPI?

lzytaro, 2021-12-21:
Hi , I want to know when will support ppi communication?

robinson, 2021-12-21:
Hi,
Thanks for asking, I am trying to actively support this library but I have limit time. 
btw, nowaday the communication with S7 less using serial, as far as I know. If you can support on this, very happy!

lzytaro, 2021-12-22:
> Hi, Thanks for asking, I am trying to actively support this library but I have limit time. btw, nowaday the communication with S7 less using serial, as far as I know. If you can support on this, very happy!

I have used this library to connect smart200 and s7-1200, and the PPI port of s7-200 still needs to be supported

robinson, 2022-01-04:
> I have used this library to connect smart200 and s7-1200, and the PPI port of s7-200 still needs to be supported

Hi, sorry for the lately response, I was busy. I have seen, it is not a rocket science in order to implement PPI, since the serial connection already implemented, see: https://github.com/robinson/gos7/blob/master/serial.go , but actually I am really busy at this time, at lease until Mar, have you take a look in it? if could you give some idea on it, woule be very helpful.