robinson/gos7 #59: When playing the example from the README, I get the following error binary.Write failed: binary.Write: invalid type int and ISO : Invalid Buffer passed to Send/Receive'19'

warleyjr10, 2023-04-19:
When playing the example from the README, I get the following error:

**"binary.Write failed: binary.Write: invalid type int"**

The only command that is working is the cpu information.
Details: I'm using S7-PLCSIM Advanced V5.0 simulator together with TIA Porta V18
Simulated CPU Model: 1511C-1 PN
Below the output of the log:

tcp: 2023/04/19 18:35:36 s7: sending 03 00 00 16 11 e0 00 00 00 01 00 c0 01 0a c1 02 01 00 c2 02 01 01
tcp: 2023/04/19 18:35:36 s7: received 03 00 00 16 11 d0 00 01 00 06 00 c0 01 0a c1 02 01 00 c2 02 01 01
tcp: 2023/04/19 18:35:36 s7: sending 03 00 00 19 02 f0 80 32 01 00 00 04 00 00 08 00 00 f0 00 00 01 00 01 01 e0
tcp: 2023/04/19 18:35:36 s7: received 03 00 00 1b 02 f0 80 32 03 00 00 04 00 00 08 00 00 00 00 f0 00 00 01 00 01 01 e0
**binary.Write failed: binary.Write: invalid type int**
tcp: 2023/04/19 18:35:45 s7: sending 03 00 00 25 02 f0 80 32 01 00 00 05 00 00 0e 00 06 05 01 12 0a 10 02 00 02 0a 96 84 00 00 40 00 04 00 10 00 00
tcp: 2023/04/19 18:35:46 s7: received 03 00 00 13 02 f0 80 32 02 00 00 05 00 00 00 00 00 81 04
tcp: 2023/04/19 18:35:47 s7: sending 03 00 00 1f 02 f0 80 32 01 00 00 05 00 00 0e 00 00 04 01 12 0a 10 02 00 02 0a 96 84 00 00 40
tcp: 2023/04/19 18:35:47 s7: received 03 00 00 13 02 f0 80 32 02 00 00 05 00 00 00 00 00 81 04
**ISO : Invalid Buffer passed to Send/Receive'19'**

hongjinlin, 2023-04-19:
Hi, please check the following 2 points.
1. DB must not be optimized.
2. GET/PUT must be enabled

warleyjr10, 2023-04-20:
Hi, I checked the settings but I'm getting the same errors.
[image]
[image]

hongjinlin, 2023-04-20:
Maybe there is a difference between the real hardware and the simulator, I encountered the problem "ISO : Invalid Buffer passed to Send/Receive '19'" in the S7-1200, which was solved by checking these 2 items

warleyjr10, 2023-04-20:
Are you also using simulator?

jiekechoo, 2023-04-23:
I use PLC S7-300, 400, 1200, 1500,  `gos7` runs  good. I guess you should use real hardware device. @warleyjr10

warleyjr10, 2023-04-24:
Thanks for the answers, I will be testing on real hardware devices