robinson/gos7 #60: How connect s7 smart200 plc

Ri0nGo, 2023-06-15:
hi,I can't to connect s7 smart200 plc.
This my code:

```
func main() {
	handler := gos7.NewTCPClientHandler("192.168.2.123", 2, 0)
	handler.Timeout = 10 * time.Second
	err := handler.Connect()
	if err != nil {
		log.Printf("connect failed, device error: %s\n", err)
		panic(err)
	}
}
```

**error info：**
2023/06/15 19:38:16 *net.OpError read tcp 192.168.1.36:62273->192.168.2.123:102: wsarecv: An existing connection was forcibly closed by the remote host.
2023/06/15 19:38:16 connect failed, device error: ISO : Invalid PDU received
panic: ISO : Invalid PDU received

Using other client connections can succeed. May I ask if it was written incorrectly？
Looking forward to your reply, thanks.

Ri0nGo, 2023-06-19:
I konw how to connect smart200

closed 2023-06-19 by Ri0nGo

iicanf, 2023-09-05:
@Ri0nGo same error, how to connect?

Ri0nGo, 2023-09-05:
修改源码
```go
// NewTCPClientHandlerWithConnType allocates a new TCPClientHandler By Connection type
func NewTCPClientHandlerWithConnType(address string, rack int, slot int, ct string) *TCPClientHandler {
	h := &TCPClientHandler{}
	h.Address = address
	h.Timeout = tcpTimeout
	h.IdleTimeout = tcpIdleTimeout
	h.ConnectionType = connectionTypeOP
	//h.ConnectionType = connectionTypePG // Connect to the PLC as a PG

	remoteTSAP := uint16(h.ConnectionType)<<8 + (uint16(rack) * 0x20) + uint16(slot)
	h.setConnectionParameters(address, 0x0100, remoteTSAP)
	return h
}
```

iicanf, 2023-09-06:
@Ri0nGo 谢谢，好使

SaltySooda, 2024-04-22:
same issue！that works！
ps：
you also have to make sure that rack and slot number is right（gos7.NewTCPClientHandlerWithConnectType("127.0.0.1", 0, 0, 2)），i use 0，0 for new smart200 plc。
you will get   **ISO : Invalid PDU received** if you set that wrong。