robinson/gos7 #68: Suggestion:  ADD new func "NewTCPClientHandlerWithConnectType" to change connectionType when allocates a new TCPClientHandler

Hongliukai, 2024-03-14:
I noticed that using NewTCPClientHandler(address string, rack int, slot int) to create a TCPClientHandler limits the connectType to 1 (AS PG_OR_PC)
```
func NewTCPClientHandler(address string, rack int, slot int) *TCPClientHandler {
	h := &TCPClientHandler{}
	h.Address = address
	h.Timeout = tcpTimeout
	h.IdleTimeout = tcpIdleTimeout
	h.ConnectionType = connectionTypePG // Connect to the PLC as a PG
	remoteTSAP := uint16(h.ConnectionType)<<8 + (uint16(rack) * 0x20) + uint16(slot)
	h.setConnectionParameters(address, 0x0100, remoteTSAP)
	return h
}
```
I often encountered when using gos7 to connect to Siemens PLCs that the PLC supports fewer PG connections and more OP connections, so I'd like to add a parameter to change this connect parameter.
```
// NewTCPClientHandlerWithConnectType allocates a new TCPClientHandler with connection type.
func NewTCPClientHandlerWithConnectType(address string, rack int, slot int, connectType int) *TCPClientHandler {
	h := &TCPClientHandler{}
	h.Address = address
	h.Timeout = tcpTimeout
	h.IdleTimeout = tcpIdleTimeout
	h.ConnectionType = connectType
	remoteTSAP := uint16(h.ConnectionType)<<8 + (uint16(rack) * 0x20) + uint16(slot)
	h.setConnectionParameters(address, 0x0100, remoteTSAP)
	return h
}
```

Hongliukai, 2024-03-14:
#67

robinson, 2024-03-15:
thanks for the PR!

closed 2024-03-15 by robinson