robinson/gos7 #75: AGReadMulti - limited to 20 items

led0nk, 2025-04-22:
Hey, 
i recently stumbled across this here:

https://github.com/robinson/gos7/blob/7ea1d6fb9d204138a117f1255763ad7b37a05e33/multi.go#L140-L147

Is there any reason for this limitation of max. amount of items set to 20?

I'm curious if this is due to some cpu-limitation of Siemens, or if it's possible to increase this.
Otherwise i would create a PR for this. And if you don't want that i need to build a wrapper at our repository.
But therefore i would love to get some experience by yours.

led0nk, 2025-04-23:
okay thanks, would be very nice to know also if this is some time-dependant multiRead, otherwise I'm thinking of doing queued requests there.
Let me know if you find sth there

younes199511, 2025-08-28:
Hello @robinson 
Is it possible to make this limit configurable by the client with default at 20 ?
Thanks